yujinrobot/yujin_ocs #73: replace table with waypoint

jihoonl, 2015-01-12:
[Table.msg](https://github.com/yujinrobot/yocs_msgs/blob/indigo-devel/msg/Table.msg) has been used to indicate a location with name.

After discussion with @bit-pirate, Table.msg will be merged with [Waypoint.msg](https://github.com/yujinrobot/yocs_msgs/blob/indigo-devel/msg/Waypoint.msg)

jihoonl, 2015-01-12:
* yujin_ocs : https://github.com/yujinrobot/yujin_ocs/tree/table_cleanup
* rocon_qt_gui : https://github.com/robotics-in-concert/rocon_qt_gui/tree/table_cleanup
* rocon_demos
 * https://github.com/robotics-in-concert/rocon_demos/tree/table_cleanup
 * https://github.com/robotics-in-concert/rocon_demos/tree/table_cleanup_rapp

jihoonl, 2015-01-12:
Table -> Waypoint converter has been developed. https://github.com/robotics-in-concert/rocon_demos/blob/table_cleanup/scripts/concert_table_to_waypoint.py

**Remain job**
* [ ] update authoring tool to use waypoint msg
* [ ] convert data in xavier and alien
* [ ] Test our demos
* [ ] Test to add and remove annotations.

dwlee, 2015-01-21:
I updated following jobs

- update authoring tool to use waypoint msg
- update webapps to parse weypoint msg
  -  https://github.com/robotics-in-concert/rocon_web_gui/pull/7
  -  https://github.com/robotics-in-concert/rocon_web_gui/pull/8
- Test our demos
  - pickup delivery test complete
  - vm delivery test complete

closed 2015-01-28 by jihoonl