yujinrobot/yujin_ocs #79: [yocs_joyop] change acceleration profile

bit-pirate, 2015-04-14:
Right now the command velocity is simply scaled by the given parameter, i.e. joystick 0 ... 1 -> 0 scale_param m/s. Using this approach it is hard to tune this parameter to support both small velocity increments and a high maximum velocity at the same time.

I'm not sure what the best/a better alternative is.

**Ideas**
 1. Use the scale parameter, but add a separate max. velocity parameter. If max vel param > scale param, then the velocity is increased by a fixed rate until max. vel is reached, when the joystick is moved to the max. position.
 2. Map joystick input to acceleration, not velocity

Other ideas? Comments? I think, I would prefer 2. Will do some testing.

jihoonl, 2015-04-15:
:+1: to 2.

bit-pirate, 2016-07-28:
I'm closing this for now, since we have a low and high-speed mode implemented (ref #99). Re-open if that's not enough and there is still a need for this.

closed 2016-07-28 by bit-pirate