yujinrobot/yujin_ocs #85: [yocs_diff_drive_controller] hard-coded factor for parameter

bit-pirate, 2015-10-20:
@AlexReimann I just ran across this:

```
[...]
dist_eps_ = dist_eps_ * 0.2;
[...]
orient_eps_ = orient_thres_ * 0.2
```
https://github.com/yujinrobot/yujin_ocs/blob/indigo-devel/yocs_diff_drive_pose_controller/src/diff_drive_pose_controller_ros.cpp#L113-L129

What is the reason for (was the motivation behind) using this hard-coded factor?

bit-pirate, 2015-10-20:
@stonier FYI

AlexReimann, 2015-10-20:
Was already there before I touched the code, question back to you ;)

bit-pirate, 2015-10-21:
Right, GitHub's blame was misleading. :smile: 

This factor got introduced here: https://github.com/yujinrobot/yujin_ocs/commit/fb26d5a492e46b4b0a09b640596511226e31d8f8

Will think more about it and remove if I cannot come up with a good reason for it.