yujinrobot/yujin_ocs #106: [joyop & smoother] brake vs. slowing down

bit-pirate, 2016-07-26:
Currently releasing the analog stick (bounces back to zero position) and releasing the deadman button leads to the same zero-velocity output and hence is treated the same on the velocity smoother side. However, in the first case we like the robot to slow down gently, whereas in the latter we like the robot to brake and stop as fast as possible.

Let's discuss how this could be implemented best. Some existing thoughts to seed the discussion:

* Service for the smoother to change the deceleration factor during run time.
* Use different deceleration factors for those two situations.
* Use a second button to indicate brake vs. slowing down.

bit-pirate, 2016-07-26:
Before diving into implementation details, I like to review the **desired** behaviour:

1. Deadman button released -> immediate stop (safety feature)
2. Deadman button pressed & stick at zero-position
 1. User released the joystick -> slow down
 2. User took over control from navigation -> full stop
3. Optional: Brake button pressed -> full stop (safety feature)

I would like to avoid adding more buttons, since we are already running low one them (and the high-speed mode (#99) will use another one). However after thinking about the above cases I don't know how to implement it cleanly without (case 1. vs. 2.i. above).

Regarding the implementation I'm considering add a second publisher to joyop, which sends out a message when

1. The deadman button has been released after being pressed. **OR**
2. The brake button is pressed.

Thoughts?

bit-pirate, 2016-07-26:
Regarding that publisher, for Gophers we could either use the emergency stop (an empty message sub) or another cmd vel publisher, which sends its message to a "unsmoothed" topic.

I will put up an implementation for the latter, since it should be more general.

AlexReimann, 2016-07-27:
The problem with the clean implementation is that if there is navigation running and you want to do a full stop, it will only full stop for a little bit before the navigation might kick in?

bit-pirate, 2016-07-27:
If you want to force a full stop during navigation you would need to keep the deadman pressed to take over control and then the brake until its stops. Navigation will not be able to take over again until the deadman is released.

If you just press the deadman (as we do currently) the robot will slowly come to a halt (as defined by the deceleration factor).

All this assuming the user is not touching the joystick.

AlexReimann, 2016-07-27:
I don't like that.

Is there a problem with the brake always being pushable without the deadman?

bit-pirate, 2016-07-28:
Add your suggestion. Brake can now always be triggered - independent of the motors being enabled or the deadman being pressed.

bit-pirate, 2016-07-28:
Implemented by f23f009f878cd93c2b6b7d9c86726c9b04731a70.

PR #107 (2016-07-28): Joyop turbo & brake
Implements both the turbo (fixes #99) and brake mode (fixes #106).
closed 2016-07-28 by bit-pirate