yujinrobot/yujin_ocs #110: OdometryHelper when odom not yet available

AlexReimann, 2016-08-18:
The OdometryHelper currently [gives](https://github.com/yujinrobot/yujin_ocs/blob/kinetic/yocs_navi_toolkit/src/lib/odometry_helper.cpp#L38) garbage when the odometry is requested when no odometry has been published yet.
This is because the [odometry_ variable](https://github.com/yujinrobot/yujin_ocs/blob/kinetic/yocs_navi_toolkit/include/yocs_navi_toolkit/odometry_helper.hpp#L71) [is not initialized](https://github.com/yujinrobot/yujin_ocs/blob/kinetic/yocs_navi_toolkit/src/lib/odometry_helper.cpp#L21).

So should properly initialize it or throw an error etc. 
There is actually not a really a reasonable default initialization for that variable (but zero is better than nans etc.)

AlexReimann, 2016-08-18:
fyi @bit-pirate

AlexReimann, 2016-09-20:
Updated the description.

The problem mainly is that our trajectory controller uses the current odom and the past issued commands to estimate where the robot will be in the future.
This estimate is then passed to the local planner which leads to the local planner running into problems because the pose is faulty (a non-normalized quaternion as orientation)

AlexReimann, 2016-09-20:
I'll catch this for now by checking for a non-normalized quaternion.
Maybe this is the actual intent behind the default quaternion constructor leading to an invalid quaternion (or rather not normalized).
Maybe close this..?

stonier, 2016-09-25:
Sending out a non-normalised quaternion and then requiring a user to *interpret* that as uninitialised would be quite bad design. There are much easier ways to help a user understand its uninitialised than that.

Initialising it as well with `[0 0 0 1]` would also be quite bad design. There are valid use cases where the user would need to know if it's initialised or not, and that is not enough information for those cases.

I'd say the cleanest solution is to have the odometry in a std shared pointer which only gets initialised after it's got data (user can check the shared pointer himself) and then move the position/velocity processing methods to library functions. Separate data collection and data processing concerns.

Otherwise use exceptions in each method for when it's uninitialised. Or a combination of the shared pointer/exception idea.

Keep in mind that exceptions are usually an expensive option. For control systems I typically only take advantage of their usefulness in constructor or init functions.

stonier, 2016-09-25:
I don't think there is anything special about these classes - I just pulled them from your clrrt just after you finished your master's period so I could re-use them in various places.

Feel free to take ownership of what is in this package.

AlexReimann, 2016-10-13:
Ohh, so it's my shitty code. I see..

AlexReimann, 2016-11-25:
Fixed with pr #119

closed 2016-11-25 by AlexReimann